AmberELEC/AmberELEC #3: Hotkey does not work

ToniBC, 2020-11-14:
When you reconfigure the controller, at the end it asks for the hotkey key, but this, even if you indicate it, does not work, the only way I can find to exit a game is by pressing L3 + R3 and in the Retroarch menu, go to the option of "close Retroarch).

Also in Retroarch, even if you modify the controls or change L3 + R3 for Select + Start, the system always forces the previous configuration and does not let you.

fewtarius, 2020-11-14:
Please use the templates for reporting issues.

closed 2020-11-14 by fewtarius

ToniBC, 2020-11-14:
Really?

What else do you want me to say, I just need to say that it is 1.0, nothing more. I cannot put screenshots because the error is of concept or of functions, not visual.

Explain the problem in detail and you surely know what I mean.

Templates are fine when it's a generic multi-platform system, but here it's unique to the RG351p.

I am telling you that the Emuelec hotkey, which is used to load and save, show the Retroarch menu, etc ... does not work, when you configure the command, the last of all is the hotkey for the extra functions, you tell it that it is the button Select and it does nothing, as I told you it is still L3 and R3.

Closing a bug report because you want a no-nonsense structure for these types of bugs seems silly to me.