AmberELEC/AmberELEC #4: Hotkey does not work

ToniBC, 2020-11-14:
# BUG
When you reconfigure the controller, at the end it asks for the hotkey key, but this, even if you indicate it, does not work, the only way I can find to exit a game is by pressing L3 + R3 and in the Retroarch menu, go to the option of "close Retroarch).

Also in Retroarch, even if you modify the controls or change L3 + R3 for Select + Start, the system always forces the previous configuration and does not let you.

# To Reproduce

- You enter the option to configure the controls.
- You press the required buttons.
- The last of all is the hotkey, I just assign Select.
- Everything remains configured, but within the games it does not work.
- When you press Select + Start or any button, functions such as Save game, Load, Exit, etc ... are usually assigned in Emuelec/Batocera / Retropie / etc ...

### Expected behavior
The hotkey would have to perform various quick functions that it does not.

### Screenshots
The error cannot be represented in screenshots. xD

### Device (please complete the following information):
 - The version is 351ELEC 1.0

### Additional context
Retroarch's generic configuration is forced on keys and certain options, if you enter retroarch and change that instead of L3 + R3 to get the menu to Select + Start, those settings are not saved. From the ES options, it should be possible to enable these shortcuts with the hotkeys and other options such as exiting the games or entering the Retroarch menu.

PD: You like it better like this, it is repeating the same thing, it seems absurd for these types of errors that images or more data cannot be provided.

Thank you.

fewtarius, 2020-11-14:
I'm unable to recreate this.  Input -> HotKeys -> Menu Toggle Gamepad Combo -> Start + Select works as expected, even after leaving retroarch and reentering it.

ToniBC, 2020-11-14:
I have tried it right now, and by magic, if it works, change the L3 + R3 to Start + Select. I think it works if you change it with a loaded game, because I tested it by going directly into Retroarch and it didn't save anything.

But the Emulect hotkeys thing doesn't work. Remember that ES (Emuelec) you have some commands for that, you press Hotkey + a button, they do different things that seem not to work here.

fewtarius, 2020-11-14:
I'll take a look, it's possible there's still a configuration missing from
the Anbernic firmware.

fewtarius, 2020-11-19:
I believe this is corrected in the maintenance images available in the #releases channel of our discord server.  Let me know if something is still awry and I'll re-open.

closed 2020-11-19 by fewtarius

jpouderoux, 2021-07-14:
Looks like it is not fixed on `351ELEC-RG351V.aarch64-20210603`. Stock hotkeys for Retroarch do not work except SELECT+START, and I am unable to configure hotkeys: pressing Function (F) key+something or Select+something just record one of the two key (either Function/Select or the chosen key). And in game, nothing happens.