AmberELEC/AmberELEC #5: The system continues to operate at 1.3 GHz

ToniBC, 2020-11-14:
### BUG
As noted in the 351ELEC 1.0 version changes, it has been patched to run at 1.5 GHz, but there is no option for that in the menu and it continues to run at 1.4 GHz.

### To Reproduce

- There is no option to put the emulators or the console at 1.5 GHz.

### Expected behavior
The most logical thing would be, that in the same way as when configuring the options of each emulator, 1.5 GHz can be selected. There is an option when you configure the emulators separately, which already puts "maximum performance" that it was supposed to be, but no.

### Screenshots
The error cannot be represented in screenshots. xD

### Device (please complete the following information):
 - The version is 351ELEC 1.0

### Additional context
Emulators such as Snes or CPS 3 games, even several of PS1, those 200 MHz more are what they need to be perfectly playable.
If it can be applied in Dreamcast and PSP, also in all other emulators.

PD: You like it better like this, it is repeating the same thing, it seems absurd for these types of errors that images or more data cannot be provided.

Thank you.

fewtarius, 2020-11-14:
The default scaler is on demand, and it's pinned to maximum performance for PSP and Dreamcast.  The next version of 351ELEC will include this information in the GUI.  You can pin any game to maximum performance or on demand in the game or system settings IU.

https://github.com/fewtarius/351ELEC/commit/b800c2a1d142a029ad7a4876fab7dc1a8dcfb273#diff-5034fb70a3f689f2d625f8acca75de922d94f82242ca7410b1f24c1beba2b2afR67

```
RG351P:~ # cat /etc/os-release
NAME="351ELEC"
VERSION="1.0"
ID="coreelec"
VERSION_ID="9.2"
PRETTY_NAME="351ELEC (Official - 351ELEC IS FREE AND NOT FOR SALE!) 1.0"
HOME_URL="https://github.com/EmuELEC"
BUG_REPORT_URL="https://github.com/fewtarius/351ELEC"
BUILD_ID="21fa50935f9e22b1c03029d6e4e62436c4b71916"
OPENELEC_ARCH="RG351P.arm"
LIBREELEC_ARCH="RG351P.arm"
COREELEC_ARCH="RG351P.arm"
COREELEC_BUILD="Official - 351ELEC IS FREE AND NOT FOR SALE!"
COREELEC_PROJECT="Rockchip"
COREELEC_DEVICE="RG351P"
RG351P:~ # cat /sys/devices/system/cpu/cpufreq/policy0/scaling_max_freq
1512000
RG351P:~ # cat /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
ondemand
RG351P:~ #
```

closed 2020-11-14 by fewtarius

ToniBC, 2020-11-14:
Thanks for the explanation. So if I understand correctly, now at 1.0, only Dreamcast and PSP run at 1.5 GHz, the rest at the normal speed of 1.3.

But in the new versions, we can put a certain game, such as Yoshis Island 2, to run at 1.5 GHz on Snes9x. The same for example with other games or emulators.

It is so. All this in the new version.

fewtarius, 2020-11-14:
1.0 is limited to ondemand as some of the upstream code the distribution is based on was broken, that's all been fixed with 1.0.1 which should release today or tomorrow.  Work is complete, just needs testing.

ToniBC, 2020-11-15:
I am testing version 1.0.1 and the processor frequencies come out, 1.3 and 1.5 of Max, but in the emulators this frequency cannot be selected. The only option is that of Max Perforance, but that option only spoke to a Retroarch Video option "Threaded Video" so that's not it.

This is normal, at the moment we cannot select 1.5 in other emulators or games.

Thank you.

fewtarius, 2020-11-15:
That it's it, it will pin max cpu frequency.