AmberELEC/AmberELEC #12: EmulationStation not displaying game name from Gamelist.xml "name" property

anthonycaccese, 2020-11-16:
### Describe the bug
After scraping metadata for a system and updating game lists the name of the game displayed in EmulationStation still displays as the filename of the game instead of the value from the "name" property in gamelist.xml.  I have tested and seen this behavior using the themes Alekfull-Solo-Horizontal, ES-Theme-EmuElecCarbon and one my own themes [ES-Theme-Artbook-3-2](https://github.com/anthonycaccese/es-theme-art-book-3-2)

### To Reproduce
Steps to reproduce the behavior:
1. Using any of the themes listed above
2. Scrape a system using the "Scrape" option from the Main Menu (in my case I am testing with GBA)
3. Scrape from Screenscraper with standard settings
3. Allow scraper to finish
4. Open Main Menu again and go to "Games Settings" > "Update Games Lists"
5. View the gamelist.xml for the system scraped compared to what displays on screen and check that expected properties are displayed

### Expected behavior
The name listed for a game in EmulationStation is expected to match the name property for that game in gamelist.xml

### Screenshots
[image]
[image]

### Device (please complete the following information):
Version 1.0.1 of 351ELEC
[image]

### Additional context
Attached is the gamelist XML I have for my GBA system for reference ([gamelist.xml.zip](https://github.com/fewtarius/351ELEC/files/5549376/gamelist.xml.zip))

I figure there may be something simple I am missing so I am still looking on my end but just in case I thought I should submit this.  

Thank you for all of your work on 351Elec, I am really enjoying how simple and clean it is.

fewtarius, 2020-11-16:
This is the default because it's my personal preference.  It can be disabled with

UI Settings -> Show Filenames in Lists -> Disable.

anthonycaccese, 2020-11-16:
nice! I'm sorry for missing that and thank you for your response.  Closing this out.

closed 2020-11-16 by anthonycaccese