AmberELEC/AmberELEC #13: Amiga games doesn't work anymore

Atamanonai, 2020-11-19:
### Describe the bug
My amiga games worked fine with 351Elec 1.0 (with P-UAE core) but doesn't work with 1.0.2. 
And if I start **Retroarch** and choose **Load Core**, I see no Commodore Amiga core in the list

### To Reproduce
1. Change core for amiga games to libretro/p-uae in emulationstation
2. Launch an amiga game
3. the screen remains black and the device must be rebooted

### Expected behavior
With 351Elec 1.0, the game starts

### Device (please complete the following information):
 - Version of 351ELEC that you are using : 1.0.2

fewtarius, 2020-11-19:
Thank you for the report, puae was unintentionally lost as a dependency during some clean up work for 1.0.2. it will be back in 1.0.3.  A pre-release image with a fix should be available some time in the next few days which will be linked in #releases in our discord when available.

fewtarius, 2020-11-19:
I found an issue with the puae package metadata that has been corrected.  I built a maintenance image that contains this fix and a few others, it should be available in #releases on our discord shortly.

closed 2020-11-19 by fewtarius

JustGuybrush, 2020-11-19:
Thanks. I've tried to join the discord but it says that the invitation link is invalid. Could you generate a new one please?

fewtarius, 2020-11-19:
Updated, didn't realize it had expired.  The new link never expires.

https://discord.gg/bmXtCt88Tz

JustGuybrush, 2020-11-19:
Thanks. I can now join the discord but I see no channel and there is a message "no text channel". Is that normal?

fewtarius, 2020-11-19:
Can you see the announcements channel?  The rules channel has an action that needs to be taken to open access to everything.