AmberELEC/AmberELEC #15: Add Pico-8 support

XargonWan, 2020-11-20:
### Describe the solution you'd like
Implement Pico-8 support as shown in this guide: https://retrogamecorps.com/2020/11/12/guide-pico-8-on-the-rg351p/

### Describe alternatives you've considered
Following the guide manually.

Cebion, 2020-11-25:
Pico-8 is now functioning in 1.0.3

mrlnnl, 2020-11-28:
Installed 351elec v1.0.3 Update2 and followed the guide on https://retrogamecorps.com/2020/11/12/guide-pico-8-on-the-rg351p/. I did the following:

- winscp connect with root/351elec
- create pico-8 folder in /storage/.config/emuelec/ports
- unpack pico-8_0.2.1b_raspi.zip to /storage/.config/emuelec/ports/pico-8 (sdl-controllers.txt is missing from pico-8_0.2.1b_raspi.zip).
- chmod 755 pico8 and pico8_dyn (winscp right click properties check execute)
- copy lexaloffle-pico-8.png from pico-8_0.2.1b_raspi.zip to /storage/.config/emuelec/ports/images as system-pico-8.png

- edit /storage/.config/emuelec/ports/gamelist.xml add

```xml
  <game>
    <path>./pico-8.sh</path>
    <name>pico-8</name>
    <desc>The Pico-8 is a virtual machine and game engine created by Lexaloffle Games. It is designed to mimic a "fantasy video game console," by emulating the harsh hardware limitations of the video game consoles around the 1980s.</desc>
    <image>./images/system-pico-8.png</image>
    <thumbnail>./images/system-pico-8.png</thumbnail>
    <developer>Lexaloffle Games</developer>
    <lang>en</lang>
  </game>
```

- created pico-8.sh in /storage/.config/emuelec/ports/pico-8.sh with the following code:

```bash
#!/bin/sh
. /etc/profile
/storage/.config/emuelec/ports/pico-8/pico8_dyn -splore -home -root_path /storage/roms/pico-8 -joystick 0
```

- download https://retrogamecorps.files.wordpress.com/2020/11/sdl_controllers.zip unpack to /storage/.config/emuelec/ports/pico-8

- create folder /storage/roms/pico-8 and uploaded some roms

Upon launching pico-8 i get a guru meditation error.

fewtarius, 2020-11-28:
If you're using 64bit 351ELEC you need to patch the pico8 binary for it to execute and you need to use a 32bit library path.  It's a known issue and something expected to be resolved in 1.0.4 (it should automatically patch).

Here is a temporary workaround:
**Patch**
```
patchelf --set-interpreter /usr/lib32/ld-linux-armhf.so.3 pico8_dyn
```
**Execute**
```
LD_LIBRARY_PATH=/usr/lib32 /storage/.config/emuelec/ports/pico-8/pico8_dyn
```

mrlnmumc, 2020-11-28:
thnx, running patchelf fixed the issue for me.

mrlnnl, 2020-11-28:
thnx, running patchelf fixed the issue for me.

fewtarius, 2020-12-01:
This is completed, and will be included in the nightly builds to become 1.0.4.  To use, copy the files from the pico-8 package to the /ports/pico-8 folder and games to the /pico-8 folder of the FAT volume.  The start script will patch pico-8 automatically if necessary.

closed 2020-12-01 by fewtarius