AmberELEC/AmberELEC #16: [Question] How to go back to EMulationStation menu from RetroArch games?

vanfanel, 2020-11-21:
Hi @fewtarius 

First of all, thanks for this distribution, it works great on my new RG351P!!

However, I can't find the button or button combo to exit from RetroArch to the EmulationStation menu. I guess it should be as easy as assigning the "Settings->Input->Hotkeys->Quit Retroarch" button, but the included RetroArch does NOT show the "Quit Retroarch" option in "Hotkeys", for some reason.

What am I missing?

Thanks!

fewtarius, 2020-11-22:
The issues system isn't for technical support.  Use the discord link in the readme for help with 351ELEC.

closed 2020-11-22 by fewtarius

vanfanel, 2020-11-22:
@fewtarius Oh! Ok, sorry :)

vanfanel, 2020-11-22:
@fewtarius I have tried joining the channel and I get: "you do not have permission to send messages in this channel".
I have never done anything to be, well, banned from discord :(

fewtarius, 2020-11-22:
The rules channel provides instructions on how to gain access.