AmberELEC/AmberELEC #17: PPSSPP v1.10.3 - Resolution Bug & Unable to Fix

basshunter420, 2020-11-22:
### Describe the bug
PPSSPP layout resolution is too big making it impossible for everything to show on screen. This wouldn't be an issue if the Display layout editor under Graphics worked properly, though nothing seems to allow for selecting boxes within Display layout editor to even do this. Aforementioned resolution issue means button mappings impossible to change. Boxed firmware has this resolution set correctly and also allows for navigating the previously mentioned menu.

### To Reproduce
1. Open any PSP game
2. Click R-Stick to open PPSSPP
3. Go to Graphics
4. Scroll to Features
5. Open Display layout editor

### Expected behavior
Expect to be able to change Display layout editor to be able to modify displayed PPSSPP resolution to match the native RG351p resolution as stock firmware does. Being unable to do this leads to no way in properly editing control mappings etc. in PPSSPP

### Screenshots
//Stock firmware
- Example of PPSSPP setting screen:
https://i.imgur.com/chyDCpC.jpg
- Example of control mapping menu:
https://i.imgur.com/iW9i5Nd.jpg

//351ELEC v1.0.2 64bit firmware
- Example of PPSSPP setting screen:
https://i.imgur.com/HpDe3kY.jpg
- Example of control mapping menu:
https://i.imgur.com/AfQeOiB.jpg

### Device:
 - RG351p without wifi - 351ELEC v1.0.2 64bit

fewtarius, 2020-11-22:
This is due to the window size patch used to improve PPSSPP performance, it's an unfortunate tradeoff but a necessary one.

basshunter420, 2020-11-22:
I see, the performance improvement is absolutely noticeable, Jeanne D'arc was essentially unplayable on stock. Thanks for the speedy response!

closed 2020-11-22 by basshunter420