AmberELEC/AmberELEC #19: DevilutionX (Diablo port) doesn’t start.

MonsieurDaz, 2020-11-23:
### Describe the bug
Diablo doesn’t start, resulting in a black screen.

### To Reproduce
Steps to reproduce the behavior:
1. Go to 'Ports'
2. Click on 'Diablo'
3. See error

### Expected behavior
The game should start.

### Screenshots
Do I really need to take a screenshot of a black screen?

### Device (please complete the following information):
 - 351ELEC 1.0.2

### Additional context
The data file is correctly placed in the correct folder, and actually works on EmuELEC on my OGA.

fewtarius, 2020-11-23:
Ports were broken in 1.0.2.  They're hopefully fixed in 1.0.3.

MonsieurDaz, 2020-11-23:
> Ports were broken in 1.0.2. They're hopefully fixed in 1.0.3.

Weird, Doom, Quake and Outrun work.

vbuciov, 2020-12-08:
I had same issue with Diablo and Opentyrian so I modified the execution script for both games.

Diablo.sh
./${PORT} >> $EE_LOG 2 > &1 for ${PORT} > &1

Opentyrian.sh
${PORT} -t /storage/roms/ports/opentyrian >> $EE_LOG 2>&1 for ${PORT} -t /storage/roms/ports/opentyrian >&1

I'm not sure why but the >> operator is causing an issue to start the game

Archidelic, 2020-12-11:
> I had same issue with Diablo and Opentyrian so I modified the execution script for both games.
> 
> Diablo.sh
> ./${PORT} >> $EE_LOG 2 > &1 for ${PORT} > &1
> 
> Opentyrian.sh
> ${PORT} -t /storage/roms/ports/opentyrian >> $EE_LOG 2>&1 for ${PORT} -t /storage/roms/ports/opentyrian >&1
> 
> I'm not sure why but the >> operator is causing an issue to start the game besides be careful with the case sensitive on file names.

Tried this and worked, but when i restart the rg351p, both scripts go back to normal :(

vbuciov, 2020-12-11:
> Tried this and worked, but when i restart the rg351p, both scripts go back to normal :(

you´re right for some reason it goes back to normal once you restart the console. I haven´t figure out how to solve it.

fewtarius, 2020-12-18:
Want to confirm this is fixed in a nightly build before closing.

overburner, 2020-12-24:
Рersonally, the DevilutionX port works fine in 1.0.5

fewtarius, 2020-12-24:
Alright, thanks for the info.  Closing this one.

closed 2020-12-24 by fewtarius