AmberELEC/AmberELEC #35: Where to find the 351Elec logo for theme creators?

nicosensei, 2020-12-05:
Hi,

I have a couple themes I made for EmuElec on the Ordoird Go Advance, that I'd like to move over to 351Elec / RG351P. Basically the only modification I'd do would be replacing the EmuElec logo by the RG551 logo, but I can't find it anywhere...

The image displayed on boot would do, but as far as I know it's a base64 encoded image in the kernel image itself.

Good job on this distro!

fewtarius, 2020-12-05:
Thank you!  The logos can be found here: https://github.com/fewtarius/351ELEC/tree/main/distributions/351ELEC/splash

closed 2020-12-06 by fewtarius

nicosensei, 2020-12-06:
Great! Many thanks!

fewtarius, 2020-12-06:
> Great! Many thanks!

No problem!  Send me the repos of your themes and I'll add them to the theme list. :)