AmberELEC/AmberELEC #43: More Scaling/Video Options in EmulationStation

myReaper, 2020-12-09:
Hi,

as written in Discord here the Feature Request to add the RGA Scaling and Bilinear Filtering Option to the Per System Configuration in Emulation Station.

Thank you :-)

fewtarius, 2020-12-09:
Please use the templates to open feature requests.

myReaper, 2020-12-09:
Updated

myReaper, 2020-12-10:
Another update:
The "Billinear Filtering" option in the RetroArch seems to be called **video_smooth = "true"** in the config file and there is an option "Smooth Games" in EmulationStation which probably are all the same. Maybe its an option to simply rename the option in EmulationStation "Smooth Games / Bilinear Filtering"  (like both names) to avoid any confusion.

fewtarius, 2020-12-18:
This is finished, and will be available in tonight's nightly build.

closed 2020-12-18 by fewtarius