AmberELEC/AmberELEC #49: Star Wars Episode 1: Racer don't work well. N64 & Dreamcast versions.

Zlitus, 2020-12-15:
### Describe the bug
Hello, I'm trying with every OS available on the RG351P to launch Star Wars Episode 1: Racer, on N64 or on Dreamcast. On N64, it launch and we can play, but we see only the bottom-half of the screen in game, like if the camera was too high compare to it original position.
On Dreamcast, we can launch the game but it is very laggy and it freeze on the menu before to reach the track.
### Device (please complete the following information):
I'm using 315ELEC 1.0.4

Zlitus, 2020-12-16:
I found a solution by using the RetroArch options and by using another emulator for the N64 version. The problem is still there for the Dreamcast version, I'll do more investigations tomorrow

closed 2020-12-16 by Zlitus

fewtarius, 2020-12-19:
You'll find that a lot of N64 and Dreamcast games don't play well on this handheld, as they require a higher spec for emulation.