AmberELEC/AmberELEC #58: Rumble issue (I think)

Drilldozer-fan, 2020-12-25:
### Describe the bug
Whenever I try to play Drilldozer for GBA, when I get a couple minutes into the game, it crashes, and not only the game crashes, but I can't even go into Retroarch, the reset button however still works. I know that rumble was introduced in version 1.0.5 (the current version) and I think this is the issue, as I had no problem playing Drilldozer on any of the previous versions.

### To Reproduce
Steps to reproduce the behavior:
1. Go to GameBoy Advance
2. Start Drilldozer
3. Play a couple of minutes (drilling triggers the rumbling)
4. It crashes (maybe it's just my unit)

### Expected behavior
I want the game to just play normally and not crash, I'd even turn off the rumble if that really is the issue.

### Device: RG351P
 - Version 1.0.5 of 351ELEC

taquitoboy, 2020-12-26:
Maybe this has something to do with the safe overclock? It runs fine on 1.0.4 but crashes frequently on 1.0.5

fewtarius, 2020-12-26:
There is no safe overclock, that was pulled as it wasn't as stable as we expected.  I was able to recreate and fix this issue though, it's available in the current nightly build and will be in 1.0.6.

fewtarius, 2020-12-26:
Looks like it's not quite fixed.  I'll try and recreate it in another emu to see if it's something broken in mgba or in the rumble patch.

fewtarius, 2020-12-26:
@davidgfnet looks like we're getting a kernel panic on the rumble code.  Have you seen this in Lakka?

anthonycaccese, 2020-12-26:
If it helps I just experienced something similar playing Pokémon Pinball on GBC. Rumble On > Played for a few Mins > game hard froze > had to press reset button to reboot.

fewtarius, 2020-12-26:
I pulled the patch temporarily while we debug it.

davidgfnet, 2020-12-26:
I have not heard such thing. However there's not a lot of Lakka users.

fewtarius, 2021-02-13:
This issue has been corrected in the 2.0 branch using previous rumble methods for now.  It will be corrected with a more optimal solution in the future.  There was no negative impact to FPS in testing.

closed 2021-02-13 by fewtarius