AmberELEC/AmberELEC #61: Emulationstation - Rewind option does nothing

XargonWan, 2020-12-28:
### Describe the bug
From emulationstation I set Rewind ON and the rewind is not enabled in RetroArch. 

### To Reproduce
Steps to reproduce the behavior:
1. Enter the desired system in Emulationstation
2. Press select
3. Goto advanced system options
4. Change the rewind option from auto to off

### Expected behavior
The rewind option should be enabled in RetroArch then, but if, when the game is running, I open retroarch and I go to framthrottle menj, the rewind option is not enabled.  
Once enabled it will work as supposed with the correct hotkeys. 

### Screenshots
I don't think they are needed. 

### Device (please complete the following information):
Nightly build 351ELEC-20201223 (6febb9232a).
https://drive.google.com/drive/folders/1gt6ODZi03pPFoPIR9S_pjtuB_QJq24oJ?usp=sharing

### Additional context
Maybe we can think to auto enable the rewind feature for all the retroarch cores that are not suffering for it (like low end emulators).

XargonWan, 2020-12-28:
I tested it only on sega megadrive default core.

XargonWan, 2020-12-28:
Update: seems like in the SNES default core not only the rewind option from ES is not going to RA but even the **shader set** configuration.

XargonWan, 2020-12-28:
Updated with a summary of which system is broken where.

fewtarius, 2020-12-28:
This will be fixed in the nightly, and in the next RC.

closed 2020-12-28 by fewtarius