AmberELEC/AmberELEC #73: Use exFAT instead of FAT32

leiserfg, 2021-01-01:
### Is your feature request related to a problem? Please describe.
Since the last release, 351Elec is using fat32 for the roms folder. That's nice cause it allows windows users to write directly to the micro-ssd, but makes it slower and also has limitations on the file sizes it can handle.

### Describe the solution you'd like
Use exFAT instead of FAT32 in the roms folder. As it is faster and doesn't have the size limitations as low as FAT32. 

### Additional context
It should be possible to keep the old FAT32 format if the user doesn't want to reinstall and maybe add a tool to format it in the other case.

fewtarius, 2021-01-01:
EXFAT is not faster unless you're using kernel 5.9 or later.  It was evaluated and abandoned because it was much slower than FAT32.

See: https://www.phoronix.com/scan.php?page=article&item=exfat-linux-59&num=3

We were able to replicate these results in testing.

See: https://github.com/fewtarius/351ELEC/commit/026d5d356948e8232ac43635d8fb1b4022a1a8bc

leiserfg, 2021-01-01:
I see, using exfat will require including https://github.com/arter97/exfat-linux/ cause the kernel 351elec is using is based on 4.x.  Sorry, I should have checked that before.

fewtarius, 2021-01-01:
I'm aware of what it will require, check the commit history and you'll find that it was already added to the kernel, along with fuse-exfat, and other tools.  Without those we would not have been able to test.

fewtarius, 2021-01-01:
If I can update to 5.9 I'll definitely re-evaluate switching. :)

leiserfg, 2021-01-01:
Sorry If I sound annoying, I'm just trying to help, the repo I linked does not relay in fuse, is just a backport of the same module included in 5.7.   Maybe the issue is (at least it what happens in linux desktop) that exfat-utils keeps using the exfat-fuse by default  so one needs to patch exfat-utils with 
https://aur.archlinux.org/cgit/aur.git/tree/nofuse.patch?h=exfat-utils-nofuse 
 To make it use the in-kernel module.

fewtarius, 2021-01-01:
It's not annoying, but we've gotten a lot of requests like this lately.  I hadn't seen that patch, I'll take a look and see if it's viable for 1.0.8.  Thanks!

fewtarius, 2021-01-05:
This will be in our nightly tonight to begin testing.  So far I'm unable to re-create the bad behavior.  Thank you again for the pointer!

closed 2021-01-05 by fewtarius