AmberELEC/AmberELEC #79: Delete rom file at Game Options (SELECT Button)

xmassound, 2021-01-04:
### Getting started
It's nice to have DELETE file option within browsing roms per Platform.

### Describe the solution you'd like
Easy way would be to select (highlight) current rom and press SELECT button to show options.
Above Edit metadata, "DELETE FILE" menu will delete selected rom file.

### Describe alternatives you've considered
Toggle confirmation popup (like trashbin in windows 10) would be nice so that you can easily delete roms you
don't want.
[image]
[image]

overburner, 2021-01-04:
There are "DELETE" button inside the "EDIT THIS GAME'S METADATA" already 😄

fewtarius, 2021-01-04:
I can't guarantee that we'll add it, but I'll look into it.

fewtarius, 2021-02-03:
I started looking into this and decided that since there's a delete option on the metadata page we won't add it.

closed 2021-02-03 by fewtarius