AmberELEC/AmberELEC #87: AlekFull-Solo-Horizontal-Xargons-ED theme won't be downloaded

denisvmedia, 2021-01-08:
### Getting started
This issue existed in 1.0.5 and seems to still exist in the newer 1.0.7. I'm able to reproduce this problem on a clean installation.

### Describe the bug
I'm able to download any theme in the Theme Downloads section but not AlekFull-Solo-Horizontal-Xargons-ED. The system gets stuck on 0% and won't let you do any further downloads unless you reboot.

### To Reproduce
Steps to reproduce the behavior:
1. Go to settings (start) and go to Themes Download.
2. Scroll to `AlekFull-Solo-Horizontal-Xargons-ED` and try to download it.
3. See the download getting stuck at 0%.

### Expected behavior
I expect the theme to be downloaded without issues.

### Device (please complete the following information):
 - Firmware version 1.0.7.

### Additional context
I am able to upload the theme manually as a zip file and unpack it.

fewtarius, 2021-01-08:
This looks like a URL issue, it'll be fixed as soon as possible.

fewtarius, 2021-01-10:
This was an issue with @XargonWan's theme URL which has been corrected.

closed 2021-01-10 by fewtarius