AmberELEC/AmberELEC #88: How to get back to the old hotkeys (that we had in 1.0.5)?

denisvmedia, 2021-01-08:
I find problematic to use new RetroArch hotkeys. The major problem is selecting a save slot. Pressing select and choosing top or down is a very unlucky combination that's hard to do by one hand on the joypad we have in our devices.

fewtarius, 2021-01-08:
The community voted to adopt a standard set of hotkeys.  You can re-assign them from the RetroArch menu system, if you run into trouble setting them with a game loaded start RetroArch from the settings menu.

closed 2021-01-08 by fewtarius

denisvmedia, 2021-01-08:
Yeah, but which community? Each device has different button placement. If you have a select button, it's convenient to use it. But if you have a select button on the right, it's not good option...

fewtarius, 2021-01-08:
The 351ELEC community, we held a vote in our Discord in December.

denisvmedia, 2021-01-08:
I see. Thanks for the info. It's very strange. Maybe my fingers are wrong, but pressing select and up is extremely inconvenient. Will revert the config just for myself.