AmberELEC/AmberELEC #96: Cannot 'make world'

rayhern, 2021-01-15:
I am trying to make world in ubuntu x86_64. Here is the error I am getting. I am not sure what the issue is.

make[4]: *** [Makefile:645: libmpiwrap_arm64_linux_so-libmpiwrap.o] Error 1
make[4]: Leaving directory '/home/ray/Documents/Code/351ELEC/build.351ELEC-RG351P.aarch64-1.0.8/valgrind-3.15.0/.aarch64-libreelec-linux-gnueabi/mpi'
make[3]: *** [Makefile:841: all-recursive] Error 1
make[3]: Leaving directory '/home/ray/Documents/Code/351ELEC/build.351ELEC-RG351P.aarch64-1.0.8/valgrind-3.15.0/.aarch64-libreelec-linux-gnueabi'
make[2]: *** [Makefile:710: all] Error 2
make[2]: Leaving directory '/home/ray/Documents/Code/351ELEC/build.351ELEC-RG351P.aarch64-1.0.8/valgrind-3.15.0/.aarch64-libreelec-linux-gnueabi'
FAILURE: scripts/install valgrind has failed!
[217/367] [FAIL] install valgrind

Other than valgrind failing to install, thats all I see in the logs. Thanks guys.

fewtarius, 2021-01-15:
We require templates for bug reports and features.  For general help use discord.  The error message would have been in the back scroll, or you can use ./scripts/build valgrind for more detailed output.

closed 2021-01-15 by fewtarius

stevewallin, 2021-01-16:
see the patch here: https://bugsfiles.kde.org/attachment.cgi?id=124176

resolved issue for me

rayhern, 2021-01-18:
> see the patch here: https://bugsfiles.kde.org/attachment.cgi?id=124176
> 
> resolved issue for me

Thank you so much @stevewallin that worked perfectly.

stevewallin, 2021-01-18:
Your welcome, you may find TIC 80 and PPSSPP will likely need the git submodules cloning as they don't seem to automatically come down for me must of the time...

fewtarius, 2021-01-18:
What distribution are you building on @stevewallin?  Wondering if I need to pull this patch in.

stevewallin, 2021-01-18:
@fewtarius "Ubuntu 20.04.1 LTS (GNU/Linux 5.4.0-59-generic x86_64)" I hit this with a clean build of 1.0.8 - Looked like those constants removed in the patch were deprecated in MPI 3.0 and something updated the dependency in the tree - didn't get a chance to follow back to the source change