AmberELEC/AmberELEC #98: Not working uae4arm

overburner, 2021-01-16:
### Describe the bug
Libretro Core - uae4arm not working or it's missing.

### To Reproduce
Steps to reproduce the behavior:
1. Go to: Main menu -> Game settings -> Per system advanced configuration
2. Select: Commodore Amiga -> Emulator -> libretro/uae4arm
3. Try to start any Amiga game from Emulationstation and see default error screen.

### Expected behavior
Game should start.

### Device (please complete the following information):
 - 351ELEC v1.0.8 (Latest release)
### Additional context
It's seems, uae4arm core is missing both in the retroarch UI and in the /usr/lib/libretro

fewtarius, 2021-01-18:
Looks like it may be 32bit only, I'll look into adding it.

fewtarius, 2021-01-21:
This issue should be resolved in tonight's build.  Let me know if there's still an issue.

closed 2021-01-21 by fewtarius

overburner, 2021-01-21:
Still not working.

It looks like there is an extra check missing here: https://github.com/fewtarius/351ELEC/blob/3356f845d81ad9c85fb1b2298a4532440057c90f/packages/351elec/config/emuelec/scripts/emuelecRunEmu.sh#L346-L347
Something like this:
```
        if [[ "${PLATFORM}" == "psx" ]] || [[ "${PLATFORM}" == "n64" ]] || [[ "${PLATFORM}" == "amiga" ]]; then
                if [[ "${CORE}" == "pcsx_rearmed" ]] || [[ "${CORE}" == "parallel_n64" ]] || [[ "${CORE}" == "uae4arm" ]]; then
```

fewtarius, 2021-01-21:
oof, you're right.  Forgot about that one.

overburner, 2021-01-22:
In latest nightly (4a0ccc0) everything works perfectly 👍