AmberELEC/AmberELEC #99: Duke Nukem breaks when using the Carbon theme

FetusSandwich, 2021-01-17:
### Describe the bug
Switching to the Carbon theme from the Art Book theme and attempting to start Duke Nukem breaks eduke If you're on the Art Book theme, it works fine. When switching to the Carbon theme, it breaks the game and it will no longer load afterwards.

### To Reproduce
Steps to reproduce the behavior:

1. Switch to the Carbon theme from the Art Book theme.
2. Go to **Ports** > **Duke Nukem**.
3. Start the game.

### Expected behavior
The game loads without error.

### Device (please complete the following information):
Version 1.0.8

### Additional context
To get it to load again, you need to then switch back to the Art Book theme and restart the system in order for it to clear and function properly.

fewtarius, 2021-01-18:
I'll try to recreate it, but there may not be a lot we can do as we don't maintain this theme.

FetusSandwich, 2021-01-19:
No worries, fewt. I had just been pulling my hair out trying to figure why it wouldn't load and after several reflashes of 1.0.8 and working backwards with my QA, I was able to figure it out.

I don't doubt that you have more pressing issues to get to, but it's definitely bizarre that a UI change would cause something like this to happen and figured someone else might run into it would be worth filing.

fewtarius, 2021-01-21:
I was able to test this today, and after installing / setting the Carbon theme I'm able to start Doom just fine.  Cannot recreate.  If it's still an issue for you, try updating to the latest nightly and mention it in discord.

closed 2021-01-21 by fewtarius

fewtarius, 2021-01-21:
Oh, I confused doom and Duke Nukem.  Either way, we should work this in discord.

fewtarius, 2021-01-21:
Just to be thorough I tried and can't recreate it in Duke 3D either.