AmberELEC/AmberELEC #103: Daphne suppor

SeaWolf69, 2021-01-20:
Hello,

First of all I love everything you have done so far and it keeps getting better! 

I recently watched a YouTube tutorial done by Wagner's tech talk on "Dragon's Lair & Space Ace: Setup Daphne on the RG350" and it looked amazing! 

I looked at the supported emulators and ports page and I see no mention of Daphne emulator/ games? Is there any possibility of the RG351P running 351ELEC v1.0.8 capable of supporting Daphne emulator/ games?

Respectfully, 

SeaWolf69

DJM75, 2021-01-20:
It is already supported.

szalik-rg351, 2021-01-20:
https://www.youtube.com/watch?v=ch3NP0tvZdA

here is guide for you

fewtarius, 2021-01-21:
Closing as resolved. If you still have issues, please use discord for support.

closed 2021-01-21 by fewtarius

SeaWolf69, 2021-01-23:
Thanks for replying....I swear I checked the supported systems and hidden
systems page and there was no mention of Dapne? Anyways after watching that
tutorial I got everything up and running with no issues! Thanks again!

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>