AmberELEC/AmberELEC #105: Wifi toggle not indicating correct state

SNESFAN, 2021-01-22:
Describe the bug

Wifi indicator shows as connected with wifi enable off after exiting game or on startup.

To Reproduce

Toggle gui option for wifi enable to be in off position, it will show "not connected" for up and internet status, power off or enter a game and exit game, wifi indicator icon shows as connected, gui option for wifi enable is off but fetching a theme/update is still possible.

Expected behavior

Wifi state to be what is shown on gui toggle, and should remember last state after power cycle or leaving a game.

Device: RG351M
Version 1.0.8 nightly of 351ELEC

fewtarius, 2021-01-22:
Thanks for the report, one of our users mentioned it may be related to RetroArch.  I'll take a look and see if we can get it corrected.

fewtarius, 2021-01-22:
This has been corrected and will be available in the next nightly.  The new behavior will be for RetroArch to follow the EmulationStation WIFI state.

closed 2021-01-22 by fewtarius