AmberELEC/AmberELEC #106: Using <sortname> tag in gamelists to refine sorting order.

sub-atomic, 2021-01-23:
In Retropie, you can use `<sortname>` in gamelists so that EmulationStation ignores "The" at the beginning of game titles when sorting alphabetically. Like this:

```
<path>./Legend of Zelda, The (USA) (PRG 0).nes</path>
<name>The Legend of Zelda</name>
<sortname>Legend of Zelda</sortname>
```

The list still shows "The Legend of Zelda," but it sorts it in L instead of in T. This tag can also be used for game series with a mix of numbers and roman numerals for sequels, Pokemon games that use colors instead of numbers, etc.

Is this possible to implement in 351ELEC? Thanks.

H3X1C, 2021-01-23:
Didn't know this was a feature on retropi, but damn this annoys me more than it should I usually resort to renaming the game.

fewtarius, 2021-01-23:
I don't expect that we'll consider this due to the amount of work involved.  If someone in the community wanted to take it on, we'd certainly be happy to merge it.

closed 2021-01-23 by fewtarius