AmberELEC/AmberELEC #108: Network proxy feature support

X-Lucifer, 2021-01-23:
## Network proxy feature support
### Feature description: 
Sometimes access to github is slow, and it will take a long time to update the system or download the theme online, or even time out. Can the network settings add network proxy support? For example, manually open http or socks5 proxy to speed up access Network, thank you sincerely.

fewtarius, 2021-01-24:
Since everything uses curl, it should support the *_proxy variables.  I can't promise that I'll get them added..or when, but it's a good idea.

fewtarius, 2021-03-08:
I don't believe we'll get this one added, but you should be able to set the proxy over an ssh connection using environment variables.

closed 2021-03-08 by fewtarius

fenguoerbian, 2022-03-17:
If I ssh into the machine and use `export http_proxy="host:port"`, it does not effect the scraper, just the running ssh session. How can I set the environment variable so it will take effect for the scraper?