AmberELEC/AmberELEC #114: Some PS1 Games Controller Doesn't Work

allenhuang-sky, 2021-01-26:
### SOME PS1 GAMES UNABLE TO RESPOND TO CONTROLLER

### When I go into certain PS1 Games like "Castlevania: Symphony of the Night" or "Final Fantasy Tactics"

### To Reproduce
Steps to reproduce the behavior:
1. Start Castlevania: SOTN or Final Fantasy Tactics
2. The game loads, but I'm unable to start the game by pressing START button or any of the buttons
3. So once I get to the Castlevania title screen, I can't proceed any further
3. I checked the Input settings in RetroArch, but it doesn't seem to make a difference
4. Previously, when I was using RG351P Stock Firmware, I didn't have any of the same issues on "Castlevania: SOTN"

### Expected behavior
I should be able to press START to start playing PS1 Games

### RG351P
351ELEC Version 1.0.8

Cebion, 2021-01-26:
https://github.com/fewtarius/351ELEC/wiki/Getting-to-Know-351ELEC#playstation-default-controls 

PSX defaults to the dualshock controller which is not compatible with some early PSX games. To use the legacy PSX controller start a game and then use L3+R3 to open the menu. Select Options / Pad 1 Type / Standard. Note that this will disable rumble, as it is only supported in Dual Shock mode.

closed 2021-01-26 by Cebion

jswagner, 2021-01-26:
Is there a reason why the Standard pad isn't the default?  Forcing the dual shock emulation breaks so many games, while the standard pad works will all games.

allenhuang-sky, 2021-01-26:
> Reply to: fewtarius/351ELEC <reply@reply.github.com>

I’m not too sure, when I installed 351ELEC (v 1.08 from Github), it seemed to default to DualShock. I hadn’t specifically changed it from Standard.
But it’s a great tip, and I will be sure to always set PSX games to Standard!

Are you the creator of 351Elec Jason? It’s a great piece of firmware! 😊

Kind regards,

[cidimage001.png@01D5DC14.42BB1070]
Allen SW Huang
Senior Product Designer
M: +64 21 258 1669

Suite 8, Level 3, Textile Centre Parnell
117 – 125 St Georges Bay Road, Parnell
Auckland, New Zealand

fewtarius, 2021-01-26:
> Is there a reason why the Standard pad isn't the default? Forcing the dual shock emulation breaks so many games, while the standard pad works will all games.

@jswagner, there is a significant number of PSX games that work with the dual shock which is why it is the default.  No matter which default we set, someone will not be happy which is why we made sure switching was documented and easy to find.

@allenhuang-sky that would actually be me. :)

allenhuang-sky, 2021-01-26:
Thank you @Fewtarius<mailto:notifications@github.com>, appreciate the extra explanation! 😊

Loving 351ELEC, definitely takes my enjoyment of RG351P up a whole notch! Also looking forward to using 351ELEC for my RG351M.

Kind regards,

[cidimage001.png@01D5DC14.42BB1070]
Allen SW Huang
Senior Product Designer
M: +64 21 258 1669

Suite 8, Level 3, Textile Centre Parnell
117 – 125 St Georges Bay Road, Parnell
Auckland, New Zealand

Date: Wednesday, 27 January 2021 at 10:46 AM
Cc: Allen Huang <Allen.Huang@sky.co.nz>, Mention <mention@noreply.github.com>

@jswagner<https://aus01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fjswagner&data=04%7C01%7C%7Cedacc25ddeea44042e0808d8c243e762%7Cbceb2d94263140bc8543a5c27edba5c6%7C1%7C0%7C637472944186565605%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=DUAv%2B3%2FYFPIqIV%2BRDMixwziNPO5oCq2KrGe6Vt0X1g4%3D&reserved=0>, there are a significant number of PSX games that work with the dual shock which is why it is the default. No matter which default we set, someone will not be happy which is why we made sure switching was documented and easy to find.

jswagner, 2021-01-26:
@fewtarius I understand 'someone will be upset' but it's the difference between 'a fourth of the library's games are completely broken and unplayable until you figure out whats going on' and 'the games work but you just have to enable Dual Shock if you want to use analog and gain rumble'.  That seems like a bad default to me.

allenhuang-sky, 2021-02-15:
@Fewtarius<mailto:notifications@github.com> Just wondering if someone’s reported the Settings per System bug?

I tried changing the settings per system on the latest firmware on RG351M, but it won’t allow me to do so.

If it hasn’t been reported yet, happy to do so on Github.

Kind regards,

[cidimage001.png@01D5DC14.42BB1070]
Allen SW Huang
Senior Product Designer
M: +64 21 258 1669

Suite 8, Level 3, Textile Centre Parnell
117 – 125 St Georges Bay Road, Parnell
Auckland, New Zealand