AmberELEC/AmberELEC #115: Updating issues, i have problem updating the new version of elec straight from updating menu, it won't update, it wont extract or update anything

yellowplux, 2021-01-27:
[image]
[image]

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug
A clear and concise description of what the bug is.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - Version of 351ELEC that you are using

### Additional context
Add any other context about the problem here.

yellowplux, 2021-01-27:
I am using nightly update, it seem to do it

I was using stable, it won't

Why?

yellowplux, 2021-01-27:
This is what it look like when using stable updating 
[image]

yellowplux, 2021-01-27:
[image]

johnnyq, 2021-01-27:
yes same here not sure

fewtarius, 2021-01-27:
This is most likely just the internet being the internet.  The updates can take some time to cache.  Try again in a few hours and let us know if it's still an issue.

yellowplux, 2021-01-27:
Still an issue

vdincman, 2021-01-27:
Yes, same problem with me, it can not verify the package on stable channel, and when I try it on nighty channel, it begins the procedure but at the very end it says the package is corrupted.

yellowplux, 2021-01-27:
Yep same here

nullstalgia, 2021-01-27:
Found a solution. 

Delete the old files from `/storage/.update`

Copy the .tar and .sha into your roms folder. Rename the .tar and .sha to have v before 1.0.9 

Like this:

`351ELEC-RG351P.aarch64-v1.0.9.tar`

And copy them into /storage/.update with File Manager or an ssh terminal. (DO NOT DO IT WITH THE SD CARD IN YOUR PC! DO IT FROM THE DEVICE)

Run update like you would normally from the menu.

:)

yellowplux, 2021-01-27:
It said it failed, not updating, I follow exactly what you said

nullstalgia, 2021-01-27:
@yellowplux 

Did you also delete `update.log` ?

If so, try this:

Attempt the update normally, let it fail.

Follow the guide normally, but leave `update.log` in `.update`

And if possible, can you send a screenshot of the files in there?

yellowplux, 2021-01-27:
> @yellowplux
> 
> Did you also delete `update.log` ?
> 
> If so, try this:
> 
> Attempt the update normally, let it fail.
> 
> Follow the guide normally, but leave `update.log` in `.update`
> 
> And if possible, can you send a screenshot of the files in there?

[image]

yellowplux, 2021-01-27:
[image]

nullstalgia, 2021-01-27:
Try uploading to https://imgur.com/upload or a similar service :)

yellowplux, 2021-01-27:
[image]

yellowplux, 2021-01-27:
https://ibb.co/fDKcLcy

nullstalgia, 2021-01-27:
What are the sizes of each of the files? You can just type it out, no need for a screenshot :)

yellowplux, 2021-01-27:
Both are 9

nullstalgia, 2021-01-27:
Yep, there it is again. Make sure to download the .tar and .sha256 from your PC, then put them into the ROMs folder

fewtarius, 2021-01-27:
Looks like there was a typo in the 1.0.9 tag that caused a file not found error.  It has been corrected.

closed 2021-01-27 by fewtarius