AmberELEC/AmberELEC #116: Updating from 1.0.8 was difficult

nullstalgia, 2021-01-27:
First, doing it via the Start menu, failed as it downloaded only 9 bytes. (I did not check if it was the first 9 bytes or something else)

Update.log of one of my attempts 

```
Wed Jan 27 03:26:30 EST 2021: Network connection detected
Wed Jan 27 03:26:32 EST 2021: Update available 1.0.8 -> v1.0.9
Wed Jan 27 03:26:32 EST 2021: Fetching 351ELEC-RG351P.aarch64-v1.0.9.tar from https://github.com/fewtarius/351ELEC/releases/download/v1.0.9
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100     9  100     9    0     0     22      0 --:--:-- --:--:-- --:--:--    22
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100     9  100     9    0     0     23      0 --:--:-- --:--:-- --:--:--    23
Wed Jan 27 03:26:35 EST 2021: Verifying 351ELEC-RG351P.aarch64-v1.0.9.tar
Wed Jan 27 03:26:35 EST 2021: Verification failed: 0019dfc4b32d63c1392aa264aed2253c1e0c2fb09216f8e2cc269bbfb8bb49b5 == Not
```

Copying the .tar direct from the releases into the `.update` folder failed due to incorrect .tar magic when the .sha256 was present, and not even attempting when not present (which is inconsistent with the instructions!!!).

[image]

I was able to update by copying the .tar and sha into my roms folder, and using File Manager to copy it to `.update`

(Which, was an issue due to B selecting and A closing menus!)

Note: This was on a RG351P with a self soldered in RTL8188EUS Wifi Module, and this was from an install I did of 1.0.8 that I used lightly. (Not updated to 1.0.8)

Cebion, 2021-01-27:
Since 1.0.8 you no longer need to put them into .update but rather into roms/update/ it then copies everything into .update.

Feel free to visit us in discord for quicker help =)

nullstalgia, 2021-01-27:
Unfortunately, this is not in any documentation. 

[image]

No offense, but I feel more comfortable seeking out documentation than personal support. Especially since I have had less than stellar experiences in the past (not with your Discord, just in general with similar projects.)

fewtarius, 2021-01-27:
Looks like there was a typo in the 1.0.9 tag that caused a file not found error.  It has been corrected.

closed 2021-01-27 by fewtarius

fewtarius, 2021-01-27:
> Unfortunately, this is not in any documentation.
> 
> [image]
> 
> No offense, but I feel more comfortable seeking out documentation than personal support. Especially since I have had less than stellar experiences in the past (not with your Discord, just in general with similar projects.)

I understand how you feel, we'll make sure the documentation is updated to reflect both methods.  Our discord is a pretty friendly place if you change your mind about joining. :)