AmberELEC/AmberELEC #120: Drastic Emulator Software Failure

Klappentromba, 2021-01-28:
### Describe the bug
Everytime I try to run a NDS game in Drastic, I get an error saying that there was a software failure.

### To Reproduce
Steps to reproduce the behavior:
1. Go to Drastic
2. Click on a Game
3. See error

### Expected behavior
The screen displays a message that reads, "Software Failure. Check Log Files for more information. Guru Meditation #68B329DA9893E3409"

### Device (please complete the following information):
 Version 1.0.9 of a RG351p

fewtarius, 2021-01-28:
Drastic is not preinstalled, and requires installation from the packages menu.

Klappentromba, 2021-01-28:
Oh, I know, and I already downloaded and installed the drastic packet via the packages menu, and even tried removimg it and reinstalled it, but encountered the same error.

fewtarius, 2021-01-28:
Interesting, we tested Drastic before releasing so it definitely works.  Do you have ssh access?  I'd like to see the content of /tmp/logs/emuelec.log after launching Drastic.

Klappentromba, 2021-01-28:
I will be completely honest with you. I am very much at an entry-level here and wouldn't know how to begin trying to set-up ssh access.

fewtarius, 2021-01-28:
It's set up by default, just need to connect to WIFI.  If you're running a really recent version of Windows 10, you have ssh already.  If not, download [putty](https://www.putty.org).

At the main menu of 351ELEC (EmulationStation), press start.  Your device's IP address should appear at the bottom of the list, usually 192.168.something.something.

On your PC, open a command shell and type 'ssh root@{device IP address}'.

If you receive an error, do the same in putty using the GUI.

Once you're connected, type 'cat /tmp/logs/emuelec.log' and press enter/return.  The log should output.  Cut and paste that into a code block (<> in the comment toolbar above).

Klappentromba, 2021-01-28:
Unfortunately, the wifi network I am on is a Univeristy network that does not allow PCs and devices to share the same network. So, I am not sure I am going to be able to set-up ssh access between my PC and RG351.

fewtarius, 2021-01-28:
Alright, that's going to make it difficult to troubleshoot I'm afraid.  You might want to try a factory reset, it won't affect your games but you will need to reinstall Drastic afterwards.  If something in your drastic config is corrupted that should fix it.

Klappentromba, 2021-01-28:
Just ran a factory reset, and no dice, still getting the same error. At this point, I think it would be best to just try flashing the latest version of 351Elec onto a new micro SD card and transfer my game files on to it. I would guess that something got corrupted in the Drastic config, but given my limited skills and inability to gain ssh access, this is little hope of figuring out what. Thank you for trying to help me troubleshoot though. Cheers.

fewtarius, 2021-01-28:
Alright, let us know if that corrects the issue.  Closing for now.

closed 2021-01-28 by fewtarius