AmberELEC/AmberELEC #124: Video Shader setting doesn’t restore on load

fearedbliss, 2021-01-29:
Hello,

I installed the latest version of 351ELEC on a new RG351P and noticed that when I go into my GB or GBA emulator (doesn’t seem to matter), I can load a shader preset, save the preset for both the core and even globally, but when coming back into the emulator, the preset is set off by default. The retroarch.cfg seem to have the auto shader loading property set to true but regardless it never restores the setting properly. I tried a fresh install of ArkOS today to test this same behavior and it seems that over there the shader config does restore correctly.

The following may be related to this or may be another but if I change the setting for “Automatically save config on exit”, I can switch that from off to on (default is off), but I can’t change it back from on to off - more specifically, I can toggle it, but when I come back on a new load it it still enabled.

XargonWan, 2021-01-29:
Try to disable the RGA scaling.

Cebion, 2021-01-29:
We are aware of this issue.
You need to enable the on switch for shaders and override core options. Otherwise that enable switch setting is not enabled.
So you apply shader, save shader setting and then overwrite core settings for that toggle shaders on switch.

fearedbliss, 2021-01-29:
@XargonWan RGA scaling was already disabled.

@Cebion I forgot to mention that it doesn’t matter if I change the shader configuration inside of retroarch because one I override the shader configuration via the emulation station options (before booting the rom), if I explicitly set the shader at that point (Advanced System Settings) then that particular setting gets saved (and I assume passed into the emulator as an argument at some point) correctly. But doing it from inside the retroarch doesn’t allow the in emulator settings to stick/restore properly.

Besides that, thanks for letting me know you guys are aware of the issue :).

fewtarius, 2021-02-03:
You cannot save shaders in RetroArch as that configuration is scrubbed and set to the configuration defined in EmulationStation as soon as you start a game.  Make your changes there instead.  351ELEC 1.x had some issues with shader selection in EmulationStation that should be resolved in 2.x.

closed 2021-02-03 by fewtarius

fearedbliss, 2021-02-03:
@fewtarius Thanks for the clarification. That sounds good!