AmberELEC/AmberELEC #125: Kids Mode Hides System

lawsonsteve, 2021-01-30:
### Issue
Whenever I enable kids mode on all games it will hide certain systems, but not all.  Seems to be related to one (or possibly more) of the games xml file or related assets that was created during scraping, but not sure. For example, I have 31 Sega 32x games installed and I have edited all the games metadata via the 351ELEC UI activating kids mode.  When I did this, switching to kids mode hid the system entirely.  By trial and error I have narrowed it down to one game, WWF Raw.  Viewing the games xml file, all the paths look correct with correct file naming convention.  

### To Reproduce
Steps to reproduce the behavior:
1. Deleted the assets (images and videos) from correct folders
2. Rebooted the RG351P
3. Rescraped from Screenscrapper
4. Edited games metadata on 351ELEC UI and turned on kids mode.
5. System is now hidden in Kids Mode entirely.

### Expected behavior
Activating Kids mode on games should not hide the systems.

### XML Metadata For this game
I tried to paste here but text was truncated.  Screenshot instead..
[image]

	
### Device:
 - Version of 351ELEC - 1.0.10 (but was the same on previous versions)

### Screenshot of files and folders
[image]

[image]

[image]

lawsonsteve, 2021-01-31:
I found a work around. The issue occurs in the last game in the system.  I have made a duplicate of the game and set this one to kids mode, leaving the other set to normal.

fewtarius, 2021-01-31:
There may have been a message in the logs concerning why, it would be in /tmp/logs.

Condemned87, 2021-02-26:
The kids mode works until I restart the system.. then I get the message "all systems are hidden, systems goes back to FULL"... but 351elec should show Gameboy and SNES, as selected previously. The games were marked as kids games. Any hint to solve it? My son will get a 351p next month...

Condemned87, 2021-02-26:
> 
> 
> I found a work around. The issue occurs in the last game in the system. I have made a duplicate of the game and set this one to kids mode, leaving the other set to normal.

That works...thanks! 2 copies of each game. one marked as kids game, one normal. But maybe someone will fix this issue. Thanks!

lawsonsteve, 2021-02-27:
> That works...thanks! 2 copies of each game. one marked as kids game, one normal. But maybe someone will fix this issue. Thanks!

To be clear, only one game in the system will need to be not a kids game.  All the other games can be marked as a kids game.  No need to have 2 copies of each game.

fewtarius, 2021-03-08:
Closing this one, might be something to report upstream or may be resolved when we next merge from upstream.

closed 2021-03-08 by fewtarius