AmberELEC/AmberELEC #132: Reset es_systems.cfg issue

aangry99, 2021-02-01:
Im using 351elec 1.0.10

After copying by modifying es_systems.cfg, "storage/newcfg" is created and "es_system.cfg" is initialized to its original state.

I think 
In'.config/emuelec/config/emulstation-config 
-----------------------------------------------------
''if [[!  -f "/storage/newcfg" ]];  then
 cp -f "/usr/config/emulationstation/es_systems.cfg" "/storage/.config/emulationstation/es_systems.cfg"
 touch /storage/newcfg
 fi'.
----------------------------------------------------
I think the script runs when os starts.

 Is there any way to modify "es_systems.cfg"?

aangry99, 2021-02-01:
can i stop the above script temporarily or permanently?

fewtarius, 2021-02-01:
It's not possible to disable that functionality.  If there's a change that needs to be made, maybe we should make it in the source?  What are you trying to accomplish?

aangry99, 2021-02-01:
I had an existing setting and would like to continue using it.

fewtarius, 2021-02-01:
Let's talk about this in Discord, if we can understand what you want to change we may be able to change it globally.

closed 2021-02-01 by fewtarius