AmberELEC/AmberELEC #134: gzdoom

fabinhojcr, 2021-02-02:
today i came across the mod Castlevania: Simon's Destiny for Gzdoom and it's a great Doom mod, is it possible to add Gzdoom to 351ELEC and play this mod?

fewtarius, 2021-02-02:
As I've previously mentioned, we require use of templates for bug reports and feature requests, and discord for support.

closed 2021-02-02 by fewtarius

fabinhojcr, 2021-02-02:
I don't use discord and I didn't make any requests, I just asked the question if it is possible to use gzdoom on the system.

XargonWan, 2021-02-02:
FYI: I am trying to port lzdoom but I don't think that that mod can work on lz. 
Gz instead needs opengl that is not supported. 
But we may continue this discussion on Discord.

fabinhojcr, 2021-02-02:
@XargonWan From what it seems to me, gzdoom would also be too heavy to run on rg351p.