AmberELEC/AmberELEC #135: Brightness setting not persistent through reboot/shutdown

luminose, 2021-02-02:
### Describe the bug
Brightness setting defaults to 100%. Changing the setting and rebooting resets the brightness back to 100%

### To Reproduce
Steps to reproduce the behavior:
1. Go to Main Menu, System Settings
2. Change brightness to anything other than 100%
3. Click Back, Quit, Reboot
4. Brightness is now back to 100%

### Expected behavior
Updated brightness value should persist after reboot. 

### Device (please complete the following information):
 - v1.0.10
 - New install. Full image flashed over a previous ArkOS install using Win32DiskImager

fewtarius, 2021-02-02:
Shut the device completely off, power it back on holding the power button for a few seconds.  Test the brightness.  Be sure not to set the brightness in RetroArch.

luminose, 2021-02-02:
Thanks! I’m not sure what I was doing wrong.

closed 2021-02-02 by luminose

fewtarius, 2021-02-02:
That corrected it?  There seems to be something weird with the newer models hardware that causes brightness to not work now and then.

luminose, 2021-02-03:
Yeah. Not sure what I did differently from before though. I was doing shutdown and reboot through the system menu.

fewtarius, 2021-02-03:
It probably wasn't you, this hardware has some bugs.