AmberELEC/AmberELEC #142: v2.0 - Known Issues Tracker

fewtarius, 2021-02-05:
This issue consolidates and documents all known issues with version 2.0 so they can be tracked and resolved as we prepare for release.

# Critical
Issues that will block the release if not resolved.

# Moderate
Issues that are serious but should not block the release.

# Nice to Have
Issues that we would like to fix, but may push to a future point release.

- [ ] Vulkan support is implemented but not yet working due to an issue with Mali upstream.

anthonycaccese, 2021-02-06:
Additional findings from today to add above:
- music not working when playing the following ports: hydra castle labyrinth, vvvvvv, c-dogs (regression from 1.0.10)
- slow framerates experienced in the following ports: vvvvvv, hydra castle labyrinth (regression from 1.0.10)
- scrolling within file manager begins to slowdown after a while  (regression from 1.0.10)

anthonycaccese, 2021-02-08:
re: Sega Master System - Advanced "Game" Option overrides are not saving
- it was related to file naming.  
- My phantasy star version was patched with a re-translation and I had named it "Phantasy Star (Japan) [T+Eng2.0_Maxim]" so i knew the version number and maker of the patch from the filename.
- The "+" "." and "_" symbols were what was causing game overrides not to retain in the ES UI
- I changed the filename to simply "Phantasy Star (Japan)" and game overrides work perfectly now

anthonycaccese, 2021-02-11:
some videos cause ES to restart when scrolling through gamelists (regression from 1.x) - see: https://drive.google.com/file/d/1RBTK9gV_PnHboGgWs-yyiIA1rBkExE1k/view?usp=sharing

this is the error that shows up in es_log.txt when the scrolling to a game where the video causes ES to restart:
`2021-02-10 03:18:59    ERROR    Interrupt signal SIGSEGV received.`

And a link to a video that I can use to consistently trigger this to occur with:
https://discord.com/channels/777665344289898536/795709999870640129/808968595522191390

This video was scraped from screenscraper.fr and worked on 1.x

closed 2021-02-17 by fewtarius