AmberELEC/AmberELEC #164: How to set date time?

PolywickStudio, 2021-02-11:
### Describe the bug
How do you set Date and Time on 351Elec?

### To Reproduce
1. The date is always wrong.

Steps to reproduce the behavior:
1. When booting up, the date and time is always wrong.

### Expected behavior
1. Correct date and time. Where to update?

### Device (please complete the following information):
Version 2.0 aa6c0be

acridAxid, 2021-02-11:
I'm on 2.0 aa6c0be, my time zone is set to America/Los_Angeles (my local time zone where it is currently 1:42AM) but clock shows "10:37".  UTC is 9:42AM.

zulhailmie, 2021-02-11:
IIRC because there are no RTC chip so when the device completelydrained out or have the battery disconnected, it'll show the latest time stored.

Connecting to the internet should synchronize the time according to the time zone 👍

acridAxid, 2021-02-11:
Yes.  I thought I was when I replied, but I am now and time synchronized correctly as expected.

SmirksAlot351, 2021-02-11:
FYI, connecting to the internet is what will sync the time based on your selected time zone.

closed 2021-02-11 by SmirksAlot351