AmberELEC/AmberELEC #167: RGA scaling: scaling issue when going into the RA menu and back to game

vanfanel, 2021-02-12:
Hi!

Maybe this is a RetroArch issue and not a 351ELEC issue but just in case, I am reporting it.

-In most (all?) RetroArch cores, if RGA filter is active, going from the game to the RA menu and back to the game causes the scaling to break, resulting in only a portion of the game screen being visible.

-In GenesisPlusGX core, with MegaDrive games only, using RGA scaling at all causes the same scaling problem: only a portion of the game screen being visible.

So, if this is a RetroArch 1.9.0 issue, this can be of course closed.
It's just that RGA looks THAT good!

SmirksAlot351, 2021-02-12:
Could you please provide the steps to reproduce and the version of 351ELEC? (Preferable use the template) as I’m unable to reproduce this behavior.

vanfanel, 2021-02-13:
### Describe the bug
Using RGA scaling causes GenesisPlusGX core to show garbled graphics in MegaDrive/Genesis games: screen is stretched and deformed, and only a part of the game screen is shown.

Also, in other cores (Nestopia) going from game to RetroArch menu and back to game causes the same defect.

### To Reproduce
Steps to reproduce the behavior:
1. On the ES menu, select GenesisPlusGX as MegaDrive/Genesis core and Nestopia as NES core.
2. On the ES menu, set RGA filter to ON.
3. Load a Megadrive/Genesis game (Golden Axe will do).
4. You will see the image is stretched and incomplete (wrong scaling). 

OR

1. On the ES menu, select GenesisPlusGX as MegaDrive/Genesis core and Nestopia as NES core.
2. On the ES menu, set RGA filter to ON.
3. Load a NES game. Enter the RetroArch menu (Select + X), and select Quick Menu -> Resume
4. Watch the game scale go wrong too.

### Expected behavior

Steps to reproduce the behavior:
1. On the ES menu, select GenesisPlusGX as MegaDrive/Genesis core and Nestopia as NES core.
2. On the ES menu, set RGA filter to ON.
3. Load a Megadrive/Genesis game (Golden Axe will do).
4. Scaling is correct. 

OR

1. On the ES menu, select GenesisPlusGX as MegaDrive/Genesis core and Nestopia as NES core.
2. On the ES menu, set RGA filter to ON.
3. Load a NES game. Enter the RetroArch menu (Select + X), and select Quick Menu -> Resume
4. Game resumes normally with correct scaling as before entering the RetroArch menu.

### Device (please complete the following information):
- Device: RG351P 
- Version of 351ELEC: Latest 1.0.11

SmirksAlot351, 2021-02-13:
Thank you for taking the time to describe the steps to reproduce. I followed them and scaling for me is still as it should be. 

Would you mind joining us in the discord and we could troubleshoot your issue further?

I’m suspect your retroarch config files are in a bad state and will need to a reset which can be found in the danger zone menu.

SmirksAlot351, 2021-02-13:
Also. You didn’t specify but I’m assuming you also set the game ratio to 4/3? (just below the emulator option) which should be already set as the default setting for Genesis and NES.

vanfanel, 2021-02-13:
@SmirksAlot351 

I have ratio in ES -> GAMES SETTINGS set to "CORE PROVIDED".
And in PER SYSTEM ADVANCED CONFIGURATION, they are all set to "AUTO", so "CORE PROVIDED" is applied to them all.

About discord: I tried many times to access via the invite link on the README, only to be greeted by a message saying "You do not have permission to send messages in this channel." I am not trying anymore, sorry.

SmirksAlot351, 2021-02-13:
Please try setting PER SYSTEM ADVANCED CONFIGURATION aspect ration to 4/3 for Genesis and NES instead of auto. We set them this way by default for a reason as not all cores handle “core provided” correctly. 

As for discord you just need to read the rules and click on an emoji to accept them. This will open all the channels and grant you permissions to contribute.

vanfanel, 2021-02-13:
@SmirksAlot351 I tried setting PER SYSTEM ADVANCED CONFIGURATION aspect ration to 4/3 for Genesis and NES instead of auto, and if I enable RGA filtering, Genesis games still look wrong here (Addadms Family looks bad for example, stretched to the right so only part of the screen is visible).

Well, if you can't see the same problem then it can't be fixed: as an opensource developer I am used to that situation.
So I will simply disable RGA filtering until more people notices (the problem is there, for sure, but looks like a RetroArch problem, not something you can fix downstream in 351ELEC) and it gets fixed somehow.

Thanks!

SmirksAlot351, 2021-02-14:
Can’t reproduce this and issue may be upstream so will close this until further reports are observed.

closed 2021-02-14 by SmirksAlot351