AmberELEC/AmberELEC #168: Atari Jaguar games not showing

PGD44FOR3, 2021-02-12:
### Getting started
I am using an RG351P with 351elec version 1.0.11
### Describe the bug
My Atari Jaguar folder and roms are not showing

### To Reproduce
Steps to reproduce the behavior:
1. Rebooted
2. Recopied roms to atarijaguar folder
3. downloaded multiple bios files
4. Not showing

### Expected behavior
I put Atari Lynx roms in the correct directory and they showed fine. Atari Jaguar roms dont show

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - Version of 351ELEC is 1.0.11

### Additional context
Add any other context about the problem here.

PGD44FOR3, 2021-02-12:
Was hoping to try a few Jaguar roms, any ideas of what the poblem could be?

SmirksAlot351, 2021-02-12:
Jaguar is not a supported system as shown on our wiki.

https://github.com/351ELEC/351ELEC/wiki/Supported-Emulators-and-Ports#what-doesnt-run

closed 2021-02-12 by SmirksAlot351

PGD44FOR3, 2021-02-12:
Sorry. Missed it. 
Thank you