AmberELEC/AmberELEC #169: Screen Dimmed after turning-on 351ELEC 1.0.11

allenhuang-sky, 2021-02-13:
The RG351M screen seems to be dimmed after booting-up

**Steps to reproduce the behavior:**
1. Currently on 351ELEC 1.0.11
2. Turn-on RG351M console, initially the screen looks 100% bright
3. Once the initialisation has loaded, the screen looks a bit dimmed
4. Going into system settings, it's already set a 100% brightness
5. Boots-up RetroArch, and the screen appears dimmed
6. I navigate to Video options under RetroArch, it appears the brightness setting is 100%
7. Once I go into brightness settings the screen is actually 100% bright again
8. Tried saving the RetroArch configuration, but once I turn-off and turn-on RG351M, the dimming occurs again
9. I'm somewhat sure it's 351ELEC causing it, as initially in the boot-up process the screen is 100% bright, but goes slightly darker at the end of the initialisation screen 

**Expected behaviour**
On the previous 351ELEC firmware, this brightness issue didn't seem to exist
This issue started after my most recent firmware update a few days ago

**Device**
 - RG351M
 - 351ELEC 1.0.11

fewtarius, 2021-02-13:
This issue was introduced while correcting another bug, it's corrected in the 2.0 branch which is expected to be available as a release candidate within the next week or two.

closed 2021-02-13 by fewtarius

allenhuang-sky, 2021-02-13:
> Reply to: 351ELEC/351ELEC <reply@reply.github.com>

Ah great – looking forward to 2.0! Thank you Fewtarius 😊

Kind regards,

[cidimage001.png@01D5DC14.42BB1070]
Allen SW Huang
Senior Product Designer
M: +64 21 258 1669

Suite 8, Level 3, Textile Centre Parnell
117 – 125 St Georges Bay Road, Parnell
Auckland, New Zealand