AmberELEC/AmberELEC #175: Diablo (DevilutionX) wrong sound effects and voices

vanfanel, 2021-02-13:
### Describe the bug
NPC characters: they all seem to have a robotic voice. Sound effects seem to be affected, too. This does not seem to happen main characters voice since I can clearly hear the "I dont have a spell ready" message.

### To Reproduce
1. Launch DevilutionX sourceport
2. Listen to the sound effects in the menu when you move the cursor, they are wrong-
3. Talk to "Cain the Elder" on the village square.
4. Hear him say "ghgdhgdhg! yudysudh!"

### Expected behavior
1. Launch DevilutionX sourceport.
2. Listen to the sound effects in the menu, which sound correct.
2. Talk to "Cain the Elder" on the village square.
3. Hear him talk crearly.

### Device (please complete the following information):
Latest 351ELEC version 1.0.11

### Additional context
This problem was already known two years ago: it happens because of SDL, but it was eventually fixed in SDL2. Look here:

https://github.com/diasurgical/devilutionX/issues/230

SmirksAlot351, 2021-02-14:
Thank you for reporting. This has already been fixed in our upcoming 2.0 release and has be confirmed in our latest nightly build.

closed 2021-02-14 by SmirksAlot351

Rolen47, 2021-07-04:
This bug has re-appeared in versions 20210603. Here's a video:

https://streamable.com/r2ny4v

Is it because SDL2 was rolled back to v2.0.10 in this version of 351ELEC?