AmberELEC/AmberELEC #179: v2.1 - Tracker

fewtarius, 2021-02-14:
## Required for Release
- [ ] Investigate ability to configure a shader to Cave Story port or move to native core
- [ ] Evaluate adding standalone Mupen64Plus
- [ ] Update the package tool to add package name and description (also update ES)
- [ ] Add Tate mode to EmulationStation

# Nice to Have
- [ ] Update support for bluetooth devices
- [ ] Vulkan support is implemented but not yet working in the upstream Mali drivers.
- [ ] screenshots don’t work with Retroachievements

anthonycaccese, 2021-02-16:
Nice to have: look at adding Sonic 1,2,CD de-compilations to ports. Upstream emuelec commit for reference: https://github.com/EmuELEC/EmuELEC/commit/a744c86f7bc9e1f981435d76747eaa97f473d969

szalik-rg351, 2021-02-18:
Nice to have: Moonlight

anthonycaccese, 2021-02-23:
Nice to have: fix Saturn (which last worked in 1.0.5)

SmirksAlot351, 2021-02-25:
Certain manuals don’t display properly and only display as a white screen (example: SWM and LTTP on SNES)

SmirksAlot351, 2021-02-27:
Nice to have: add MP3 support for BGM

anthonycaccese, 2021-03-15:
Nice to have: 
update Daphne default controls and change...
select = coin
start = start
r2 = exit

fewtarius, 2021-03-19:
Will be reworked, closing for now.

closed 2021-03-19 by fewtarius