AmberELEC/AmberELEC #182: Low screen brightness on RG351M

runforlife, 2021-02-17:
**Description:**
I just installed 351ELEC 1.0.11 on my RG351M. I noticed that the 351ELEC has less maximum brightness then the stock Emuelec 3.8 firmware. I like the bright screens but the maximum brightness of 351ELEC in not enought for me. Has anyone run into the same problem?

szalik-rg351, 2021-02-17:
our main theme is a little bit darker, so that might give you this feeling. Try to download the emuelec theme and see if it look same for you "es-theme-solo-horizontal-arkos-ed" this one for example. On games probably you don't notice this behaviour i guess?

SmirksAlot351, 2021-02-17:
Please feel free to join us in Discord to discuss questions such as this.

https://github.com/351ELEC/351ELEC/wiki#getting-help

closed 2021-02-17 by SmirksAlot351