AmberELEC/AmberELEC #186: Feature Request: Tool to Delete AppleDouble Files

retrogamecorps, 2021-02-18:
### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Is your feature request related to a problem? Please describe.
Many of us are Mac users and experience the pain of having AppleDouble ("._") files that appear when transferring over files from a Mac to a 351ELEC SD card.

### Describe the solution you'd like
There are apps and Terminal scripts (dot_clean) that can be ran to clean the SD card before ejecting, but a script in the Tool section of 351ELEC might be very handy.  Not sure if it's feasible in 351ELEC, but just an idea :) Feel free to close if it's not possible.

fewtarius, 2021-02-18:
As a Mac user myself I know this pain.  It's a really easy feature to add, so I'll try and get it into 2.1.

PeterFalken, 2021-02-18:
This two line script should be able to cleanup most of the offending files.

```
find /storage/roms -type f -name '._*' -exec rm {} \; 
find /storage/roms -type f -name '.DS_*' -exec rm {} \;
```

fewtarius, 2021-02-18:
Yep, should be easy.  I'll add it to one of our existing scripts and add a menu option to EmulationStation.

PeterFalken, 2021-02-18:
For a more thorough cleanup (macOS specific directories) you could also add these lines:

```
rm -rf /storage/roms/.Spotlight-V100;
rm -rf /storage/roms/.Trashes;
rm -rf /storage/roms/.fseventsd;
```

Cebion, 2021-05-01:
Was implemented

closed 2021-05-01 by Cebion