AmberELEC/AmberELEC #193: Amstrad CPC emulator CrocoDS onscreen keyboard is set up as AZERTY instead of QWERTY so many letters which are necessary to press to load games won't work.

JCORR76, 2021-02-20:
### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug
A clear and concise description of what the bug is.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - Version of 351ELEC that you are using

### Additional context
Add any other context about the problem here.

christianhaitian, 2021-02-23:
If needed, the fix to change the keyboard from French to US is available here.  https://github.com/christianhaitian/libretro-crocods/commit/997a1f4851c53eca18c17ef753e284b3420ff871

fewtarius, 2021-02-23:
> If needed, the fix to change the keyboard from French to US is available here. [christianhaitian/libretro-crocods@997a1f4](https://github.com/christianhaitian/libretro-crocods/commit/997a1f4851c53eca18c17ef753e284b3420ff871)

Thanks Christian!

JCORR76, 2021-02-23:
Actually I was going to ask Christian because he included the CrocoDS emulator on ArkOS and he fixed the same problem as well but then I realized this was not his firmware LOL

christianhaitian, 2021-02-23:
No problem.  I help here with 351Elec where I can as well.  We try to help each other across the various distros.  It's a very supportive community here.

fewtarius, 2021-03-08:
This fix should be in the next nightly build.

closed 2021-03-08 by fewtarius