AmberELEC/AmberELEC #201: VVVVVV low FPS on 2.0

Astorek86, 2021-02-23:
Hello there, it seems that the new 2.0 had some Problems running VVVVVV on my RG350M .

On current 2.0, I seems that I got around 15 FPS all the time, makes it really hard to play.

### To Reproduce
Install 351Elec on a SD-Card (no upgrade), Eject Drive safely, start RG351M while SD-Card is connected, shut down the System properly, insert SD-Card on PC and copy "data.zip" into the "ports/vvvvvv"-Folder. Eject Drive safely, start again RG351M while SD-Card is connected, activate "Ports", try to start the Game.

Result:
Game starts, but it's immediately noticeable that the FPS is lower than expected.

### Expected behavior
On v1.0.11, everything runs fine at full speed.

### Device (please complete the following information):
 - 351ELEC v2.0, on a RG351M

fewtarius, 2021-02-23:
Go into graphics options, and toggle FPS.  It should return to normal.

Astorek86, 2021-02-23:
> Go into graphics options, and toggle FPS. It should return to normal.

I tried, and you are right! Problem solved, thank you^^.

closed 2021-02-23 by fewtarius