AmberELEC/AmberELEC #209: Sonic Robo Blast 2 and Sonic Robo Blast 2 Kart ports

thisjustin2001, 2021-02-25:
Hey! As the title suggests, I'm looking to see if ports for my two favorite open-source games, Sonic Robo Blast 2 and very similarly titled Sonic Robo Blast 2 Kart would be possible on 351Elec.

Sonic Robo Blast 2 is already up and running on ArkOS, while Srb2 Kart hasn't had such luck. It's been hard to convince some people that Sonic Robo Blast 2 and Srb2 Kart aren't the same game, but in fact two completely different games built off the same engine, kinda like the relationship between Doom 1 and Doom 2.

Sonic Robo Blast 2's current version is v2.2.8, with v2.2.9 coming soon. Srb2 Kart's current version is v1.3, with v2.0 coming soon, if that helps.

If you end up listening to my suggestion, thank you so much! If not, I still appreciate that you read what your users have to say!

szalik-rg351, 2021-02-25:
join discord and drop your suggestions here, thanks

szalik-rg351, 2021-02-25:
Join Our Community: https://discord.gg/bmXtCt88Tz

closed 2021-02-25 by szalik-rg351