AmberELEC/AmberELEC #210: DevilutionX (Diablo) port, control mappings

Teroan, 2021-02-26:
### Describe the bug

The DevilutionX/Diablo port control mappings are not set properly by default.

However I've managed to load the fixed "gamecontrollerdb.txt" file provided from the ArkOS project. Everything works except the right analog stick which is suppose simulate mouse does not move (stuck to the top left corner) while it does activate properly (button clicks).

Here is the content of the "/ports/diablo/gamecontrollerdb.txt"
`03000000091200000031000011010000,OpenSimHardware OSH PB Controller,a:b1,b:b0,x:b3,y:b2,leftshoulder:b4,rightshoulder:b5,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,leftx:a0~,lefty:a1~,guide:b12,leftstick:b8,lefttrigger:b10,rightstick:b9,back:b7,start:b6,rightx:a2,righty:a3,righttrigger:b11,platform:Linux,`

Content of the modified "/ports/diablo.sh". Changes to this file changes reset between every restart(?), Is there a way make them permanent?
```
#!/bin/bash

# SPDX-License-Identifier: GPL-2.0-or-later
# Copyright (C) 2020-present Shanti Gilbert (https://github.com/shantigilbert)

# Source predefined functions and variables
. /etc/profile

PORT="devilutionx"
# init_port binary audio(alsa. pulseaudio, default)
init_port ${PORT} pulseaudio

/usr/bin/show_splash.sh "${PORT}"

clear >/dev/console

export SDL_GAMECONTROLLERCONFIG="$(cat /storage/roms/ports/diablo/gamecontrollerdb.txt)"

jslisten set "/usr/bin/killall ${PORT}"

if [ ! -L /storage/.local/share/diasurgical/devilution/diabdat.mpq ]; then

mkdir -p /storage/.local/share/diasurgical/devilution/

ln -sf /storage/roms/ports/diablo/diabdat.mpq /storage/.local/share/diasurgical/devilution/diabdat.mpq

fi

cd /storage/roms/ports

emuelec-utils devilutionx_server "$(get_ee_setting diablo.server)"

emuelec-utils devilutionx_password "$(get_ee_setting diablo.password)"

${PORT} &>>/tmp/logs/exec.log

jslisten stop

ret_error=$?

[[ "$ret_error" != 0 ]] && ee_check_bios "Diablo"

end_port

unset SDL_GAMECONTROLLERCONFIG

exit $ret_error
```

### Expected behavior
diasurgical/devilutionX: Diablo build for modern operating systems https://github.com/diasurgical/devilutionX#controller-support

```
DevilutionX supports gamepad controls.

Default controller mappings (A/B/X/Y as in Nintendo layout, so the rightmost button is attack; A ○, B ×, X △, Y □):

    Left analog or D-Pad: move hero
    A: attack nearby enemies, talk to townspeople and merchants, pickup/place items in the inventory, OK while in main menu
    B: select spell, back while in menus
    X: pickup items, open nearby chests and doors, use item in the inventory
    Y: cast spell, delete character while in main menu
    L1: use health item from belt
    R1: use mana potion from belt
    L2: character sheet (alt: Start + L1 or ←)
    R2: inventory (alt: Start + L2 or →)
    Left analog click: toggle automap (alt: Start + ↓)
    Start + Select: game menu (alt: Start + ↑)
    Select + A/B/X/Y: Spell hotkeys
    Right analog: move automap or simulate mouse
    Right analog click: left mouse click (alt: Select + L1)
    Select + Right analog click: right mouse click (alt: Select + R1)
    Select + L2: quest log (alt: Start + Y)
    Select + R2: spell book (alt: Start + B)

For now, they can be re-mapped by changing SourceX/controls or by setting the SDL_GAMECONTROLLERCONFIG environment variable (see SDL_GameControllerDB).
```

### Device:
 Anbernic RG351P, 351ELEC v2.0

fewtarius, 2021-03-04:
The scripts are overwritten on startup, there's no facility to prevent that.  We can use your findings to implement a fix.

dhwz, 2021-05-09:
Mouse support is fixed, is this issue related to more than missing mouse support?

closed 2021-09-28 by dhwz