AmberELEC/AmberELEC #212: ScummVM - Virtual keyboard bug

Manfred282, 2021-02-26:
### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug

Scumm VM Virtual keyboard cant be used. The virtual keyboard is centered at the bottom of the screen,
but the input cursor doesnt go far enough. Only "ESC" and "F1"-"F8" can be used.

This only happens ingame. If "select" is pressed and then the keyboard, it works
But no way Monkey Island 2 can be played, because of the input dialog at the beginning.

Did test it on all LucasArts games, alle the same, keyboard didnt work

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - Version of 351ELEC that you are using

351ELEC 2.0

### Additional context
Add any other context about the problem here.

stickfight, 2021-02-28:
The issue effects the EOB games as well, as you have to type in the name of your party members as well

Teroan, 2021-02-28:
Same here. Tested various settings (graphics modes, render modes, etc.) and couldn't find workaround. Also with graphics mode set to "normal (no scaling)" the virtual keyboard appears to be completely missing (off screen?) Seems like some issue where the keyboard and the "box" in which the cursor is suppose to be locked in do not match?

stickfight, 2021-03-01:
@Teroan  The A button solution, solves the initial issue for me (nice find)

fewtarius, 2021-03-04:
> Same here. Tested various settings (graphics modes, render modes, etc.) and couldn't find workaround. Also with graphics mode set to "normal (no scaling)" the virtual keyboard appears to be completely missing (off screen?) Seems like some issue where the keyboard and the "box" in which the cursor is suppose to be locked in do not match?
> 
> UPDATE:
> 
> After messing around with the following files, virtual keyboard works fine in normal (no scaling) graphics mode. Possibly this was already working (???) or i might done something to unintentionally change something...
> 
> ```
> /roms/storage/gamedata/scummvm/extra/vkeybd_default.zip
> /roms/storage/gamedata/scummvm/extra/vkeybd_small.zip
> ```
> 
> Default contains both 320x240 and 640x480 graphics/XML. Small only has the 320x240.
> 
> UPDATE2:
> 
> Appears you can actually drag the virtual keyboard by press and hold A key in the empty space around the keyboard keys. Another solution is to modify the XML files inside those ZIP. These files probably need to be modified to better fit the RG351? This is at least workable solution...
> 
> from:
> `<keyboard initial_mode="lowercase" v_align="bottom" h_align="center">`
> to:
> `<keyboard initial_mode="lowercase" v_align="top" h_align="left">`
> 
> Also changing between graphics modes will make the keyboard not to render. You have to restart scummvm. Using OpenGL seems to load 640x480 resolution graphics/layout while in normal (no scaling) it uses the 320x240.
> 
> UPDATE3:
> Actually the OpenGL seems to use the same small layout. Also heres some more details about this from the ScummVM API page:
> 
> [ScummVM API documentation / vkeybd](https://doxygen.scummvm.org/dd/d6e/virtual-keyboard-gui_8h_source.html)

Thanks for pointing this out, we'll see what we can do to improve the experience.

anthonycaccese, 2021-05-01:
This may be fixed by a most recent update. I tested this morning and was unable to replicate. Can you try again on the latest release and add additional comments?

closed 2021-05-01 by anthonycaccese