AmberELEC/AmberELEC #213: Frontend music doesn’t play on RG351M and 2.0

PapaBear7856, 2021-02-26:
### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug
A clear and concise description of what the bug is.
Frontend music doesn’t play on RG351M with version 2.0 of 351ELEC but does play correctly on 1.0.11

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - RG351M

### Additional context
Add any other context about the problem here.

anthonycaccese, 2021-02-27:
We'll need a bit more info to help on this.  What format is your background music in?  If its mp3, at this time that is not supported and we have it on our backlog to look at.

PapaBear7856, 2021-02-27:
Yes, all are mp3. Which audio formats are currently supported in 351ELEC 2.0?

BTW, thank you to all for your contributions in making awesome gaming software!

stickfight, 2021-02-28:
Very sorry to ask but i cant find it on any of the guides or FAQ, how do start the front end music app?

PapaBear7856, 2021-02-28:
Copy the audio files to the folder labeled BGM on the GAMES partition then once in the system there is an option to turn on frontend music under the Sound settings after pressing Start.

stickfight, 2021-02-28:
Thank you very much

NetFishNReel, 2021-03-02:
I am also having the same issue with the custom frontend music not playing. Using the RG351m with 351elec 2.0. All files are in mp3 as well. Made this account to let y'all  know about it and to thank you for the great work.

figurhead71, 2021-03-02:
Converting files from MP3 to OGG in the BGM folder for now is a quick and easy workaround and what I have done to fix this issue.

dhwz, 2021-05-26:
fixed with next update

closed 2021-05-26 by dhwz