AmberELEC/AmberELEC #214: Screen Scraper crashes the OS

cyohn55, 2021-02-27:
### Getting started
Whenever I attempt to download box art/videos etc with screen scraper the download is extremely slow (1 item downloaded or less per minute)

### Describe the bug
After selecting screen scraper, toggling the other scraping options on, and starting the screen scraper downloads--the system begins downloading each item very slowly and in some instances freezes the OS. It sometimes causes a reboot automatically and other times I have to manually reboot.

### To Reproduce
Steps to reproduce the behavior:
1. Go to Main menu>Scrape>toggle all "Scrape (.......)" to on>enter login credentials>scrape now
2. Freezes either immediately or after a few minutes

### Expected behavior
Expected screen scraper to download the images.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - Version 2.0 on the RG351P

### Additional context
This isn't additional context but, I was hoping I could also ask: where are the ports located? I downloaded Half-life and a bunch of Doom and Doom 2 WADs but can't seem to find them? Thank you.

anthonycaccese, 2021-02-27:
A few questions that will help us get more context:
- Are you logged into your screenscraper.fr account when starting the scrape?
- How many games are you attempting to scrape at a time?

cyohn55, 2021-02-27:
I was initially logged into my account (I thought) But, I just deleted a character from my password and scraped it again. This time it’s scraping faster than it was before but eventually got an 'ERROR: 401'. I just went on their website to check if my password is valid and it looks like theyre experiencing server issues. And last night I was scraping around 3-400 games at once, after reducing the consoles to 5 at a time.

cyohn55, 2021-02-27:
[image]
[image]

cyohn55, 2021-02-27:
[image]

SmirksAlot351, 2021-02-27:
This appears Screenscraper server issue. Not 351ELEC.

closed 2021-02-27 by SmirksAlot351